MST-Robotics/IGVC #9: Add different control types

matthewia94, 2015-09-13:
Input: Joy messages
Output: Twist messages

Package: control
Node: teleop_control

There are multiple drive modes that might be useful for tele operated control.  The current drive mode is based on arcade drive where the y axis controls linear velocity and the x axis controls angular velocity.  It would be useful to have a tank drive as well where the y axis on two different sticks controls the velocity of one side of the robot.  In other words both sticks forward would be forward, both backwards would be backwards and opposite of each other would turn.

aswgxf, 2015-11-01:
Having issues with things, gonna wait till #41 is finished, and then try from there.

aswgxf, 2015-11-01:
Also, I am working more towards #8 than this one. This would require some other changes on top of the changes in #8.

jacobgkau, 2016-10-18:
Here's what I've figured out about this issue:
- The right and left wheel speeds are published as "right_vel" and "left_vel" in motor_controller.cpp
- Those values are calculated in the same file based on a "linear velocity" and an "angular velocity"
- The linear and angular velocities are published in remote_control.cpp
- Those velocities are taken from two joystick axises, which are configured in the controller's .yaml file (for the Xbox 360 controller, the file is 360.yaml)

So, in order to have a tank drive system, rather than having a linear velocity and angular velocity which will be converted to left and right velocity later, we just need to skip the linear/angular velocities and set the joystick axis values (for axises from two different sticks) to the left and right velocities directly.

jacobgkau, 2016-10-19:
This is a list of changes I am making to the code (for my own reference):

Renaming some existing axis names to be more generic so they can be used for several different control types, rather than being named for a specific control type:
- 360.yaml: Change "lin_vel_axis" to "left_y_axis"
- 360.yaml: Change "ang_vel_axis" to "right_x_axis"
- moga.yaml: Change "lin_vel_axis" to "left_y_axis" and "ang_vel_axis" to "right_x_axis" (for consistency)
- remote_control.cpp: Line 57: Change "config.lin_vel_axis" to "config.left_y_axis"
- remote_control.cpp: Line 58: Change "config.ang_vel_axis" to "config.right_x_axis"
- common/config.cpp: Line 59: Change 2 occurrences of "lin_vel_axis" to "left_y_axis"
- common/config.cpp: Line 60: Change 2 occurrences of "ang_vel_axis" to "right_x_axis"
- common/config.h: Lines 30, 31: Change variable names from "lin_vel_axis" and "ang_vel_axis" to "left_y_axis" and "right_x_axis"

(These global changes are needed so we can use the left Y axis for tank drive without referring to it as "linear velocity", since that won't be a sufficient description in the context of tank drive.)

Now to add the new axis:
- 360.yaml: add lines (comments?) about X_AXIS_L_STICK and Y_AXIS_R_STICK and their numbers
- 360.yaml: add line "right_y_axis: 4"
- common/config.h: Add variable declaration: int right_y_axis;
- common/config.cpp: Add line: get_int_param("/right_y_axis", config.right_y_axis);

All of these changes have to do with renaming existing joystick inputs within the code and adding new joystick inputs that we will be able to use. After these changes, arcade drive should still work like it did before. I'm not done yet! Tomorrow I will actually take the new joystick inputs and publish them in remote_control.cpp, then use them to actually set motor output in motor_controller.cpp.

Again, I know this isn't how you do GitHub, but this comment is mainly to help me keep track of what I'm doing. I will actually upload the code to GitHub tomorrow after I've finished a complete Tank Drive solution.

jacobgkau, 2016-10-20:
In order to switch between different code control modes, we need a way to use different code based on a variable. The variable could be changed when a button is pushed. Putting all of the code inside of the "void RemoteControl::joyCallback" method (located on line 46 in remote_control.cpp) in an "if" statement checking for the variable would allow us to either publish a command velocity message with linear X and angular Z (differential mode) or with left linear and right linear (tank). (A switch statement may be more appropriate and may be easier to scale to other drive modes in the future.) Also, in motor_controller.cpp, in the method "void MotorController::twistCallback" (line 42), the different calculations for right and left velocity would need to run based on the mode that is set.

For now, I am temporarily commenting out code that is only useful for differential drive (e.g. lines 44 and 45 of motor_controller.cpp), and I'm just focusing on making tank drive work. After we see if it works, I can work on a switching solution for the two drive modes.

In remote_control.cpp, I added the line "cmd_vel.linear.y = msg->axes[config.right_y_axis] * speed_modifier;" For now, I am using linear.x for left velocity and linear.y for right velocity. In the future, we may want to rename these, but I'm trying to make as few changes as possible right now.

In motor_controller.cpp, I commented out lines 44 and 45 (lines that call functions to convert linear and angular joystick velocity into left and right wheel velocity). I added two new functions, MotorController::getLeftVelTank and ::getRightVelTank, that perform the same scaling operations that were used for differential drive, but each only factoring in its respective side's linear velocity and completely ignoring angular velocity. (This scaling may need work, I'm just using what was already there for now.) In the twistCallback method, I added two new lines (replacing the commented out ones) calling the new scaling functions and setting the left and right wheel velocity to the scaled values.

So in summary, what I have done is leave the overall structure of the control package in place, but disable calculations about "linear and angular" velocity and replace them with calculations about "left linear and right linear" velocity. As of right now, tank drive should work, and differential drive should NOT work, but it should be easy to re-enable it, because I only commented out a few lines that conflicted with tank drive, and I left the rest of the differential drive code in place even though it is not being used right now.

Now I'm going to try and submit my modified code as a separate branch to the repository.

jacobgkau, 2016-10-20:
I have pushed the changed code to the "tankDrive" branch. (I think there's a way to make pushes show up here in the issue, but I don't know it yet.)