MST-Robotics/IGVC #30: Statically assign COM ports to arduino

matthewia94, 2015-10-02:
Package: control

Currently the arduinos are assigned COM ports based on which one is plugged in first.  This will be somewhat cumbersome and error prone if we unplug the motor controllers and so we should find a way to statically assign them permanent COM ports.  

A bash script to statically assign would be ideal.

This is related to #29 and #28

jerryl108, 2015-10-16:
I found an article that about doing pretty much exactly what we need:
http://hintshop.ludvig.co.nz/show/persistent-names-usb-serial-devices/

The only thing is that the Arduino IDE might need the original names to be statically assigned instead of being able to use the new linked names.

Also, I can't seem to assign more than one person to the issue.

matthewia94, 2015-10-16:
@jerryl108 Only one person can actually be assigned.  I recommend you base the assignment off of which fork you will be using.

matthewia94, 2015-11-18:
Can I please get a progress update on this.  This has been open for a long time and really needs to be closed.

jerryl108, 2015-11-21:
Ben and I wrote shell scripts for this during the computing meeting a few weeks ago. I emailed him and didn't get a response, so I just rewrote the shell scripts. They haven't been tested with an Arduino. I'll make a pull request to add them to the repo if you want.

Also, I remember that the Arduino IDE doesn't see the statically assigned names (symlinks) so you might have trouble knowing which Arduino you're pushing code to if there is more than one plugged in. I might play around some more with changing the actual names and not just making symlinks later.

https://gist.github.com/jerryl108/73a2405b193b56c8f7c8

matthewia94, 2015-11-22:
A pull request would be nice, we will have @ryanloeffelman test them if he can as I do not currently have access to an arduino.  If he can't test them we can wait until we get back to accept the request.  

Thanks!