MST-Robotics/IGVC #72: Write wiki for simulator

matthewia94, 2015-11-16:
There is a gazebo based simulator merged into the repository and it would be nice if it was documented.

matthewia94, 2015-11-16:
@MST-Robotics/moderators Could some of you please review my documentation for the simulator on the wiki and let me know if it needs to be more in depth.

ryanloeffelman, 2015-11-19:
It all looks good, right up until the end, it may be useful to put a short paragraph on how to properly publish to /cmd_vel. Or you could just tell them to open up a new terminal and run our code, to test our code. Also, when I run our code, and the simulator, how do I view specific data, like the camera image to see if I am getting what I want out of the code I am testing. Just a few question I was unsure of when doing this for myself the first time

closed 2015-11-19 by ryanloeffelman